eric-wieser/2.12-robot #39: Move CAD to hardware subfolder

eric-wieser, 2015-11-20:
@premphy, @reaganm: In order to make this easy, the following needs to happen:

* You check in all your changes
* You close solidworks
* I move the folder, and commit
* You sync/pull, and can resume work

reaganm, 2015-11-20:
I have committed/closed solidworks.

premphy, 2015-11-21:
You can move the folder, Eric. I will then pull and commit my new parts into it
Thanks.

________________________________
Chandan Sharma Subedi
Department of Mechanical Engineering
Massachusetts Institute of Technology
premphy@mit.edu | 617-608-8694

eric-wieser, 2015-11-21:
Done

closed 2015-11-21 by eric-wieser

premphy, 2015-11-21:
Great!

________________________________
Chandan Sharma Subedi
Department of Mechanical Engineering
Massachusetts Institute of Technology
premphy@mit.edu | 617-608-8694

commit 71adfc71 (2015-11-21): Move hardware to subfolder (fixes #39)