eric-wieser/2.12-robot #42: Write code to control the rail dynamixels

eric-wieser, 2015-11-20:
See `software/Dynamixel config.txt` for details on some of the complexities here

eric-wieser, 2015-11-24:
Update: we can now connect to them - the issue was that they had different baud rates, and clashing ID numbers. The one on robot-right is now id 1, and the one on robot left is id 2

eric-wieser, 2015-11-24:
Both are at a baud rate of 400000

14kirchm, 2015-11-24:
Ok, great news. I'm going to lab tomorrow to write the code and see if they are able to rotate more than 1 turn unless you were already working on this ?

eric-wieser, 2015-11-24:
they can rotate more than one turn, but we need to be careful how we configure them (torque limits, etc). I don't plan to work on this - I thought I'd just debug the connection while there was someone else having the same issue.

14kirchm, 2015-11-24:
Ok sounds good. I just need to add the torque limit, angle limits and the Resolution Divider issue to the code uploaded for lab4 correct ?

eric-wieser, 2015-11-24:
I mean, most of lab4 was inverse kinematics, which is irrelevant here. Change whichever settings you think make the lifter behave most sensibly. You might find you need to tension the belts to get them not to slip - there are some tensioning springs in the box.

14kirchm, 2015-11-24:
Ok I'll take a look

eric-wieser, 2015-12-02:
I think we're good here, software-wise

closed 2015-12-02 by eric-wieser