eric-wieser/2.12-robot #49: Find a way to put snow in dumptruck

14kirchm, 2015-11-29:
Gap between dump truck and sifter. The snow doesn't go into the truck

[image]

eric-wieser, 2015-11-29:
Cardboard? Paper?

reaganm, 2015-12-01:
Relatedly: should I re-lasercut the servo mounts to be taller so that we can increase the tilt angle? Or did we just decide to use a weight as the solution?

eric-wieser, 2015-12-02:
May as well do that, and swap the motor mounts for the MX64s as well

eric-wieser, 2015-12-02:
I think using a weight will be hard to get right
And doesn't work for the debris (which is actually heavier than what we tested with)

reaganm, 2015-12-02:
Working on this now, will be back in lab soon.