eric-wieser/2.12-robot #50: Make 6-slide presentation

eric-wieser, 2015-11-30:
By midnight tomorrow

eric-wieser, 2015-11-30:
(can be modified after initial submission)

simpri21, 2015-12-01:
are the slides just about how the robot does each of its tasks (navigating, picking up snow, dumping snow, picking up tree, clearing roof)? I can work on this

eric-wieser, 2015-12-01:
I think that's most of it, yeah

eric-wieser, 2015-12-02:
I've got a couple of photos of the sifter doing things, that I'll put on google drive in a few minutes

eric-wieser, 2015-12-02:
Photos are up, although maybe a rendering would be better. Were you not able to render the sifter in the up position?

simpri21, 2015-12-02:
Working on it now, its gonna be done very last minute

eric-wieser, 2015-12-02:
I just updated the cad. It was wrong

simpri21, 2015-12-02:
I'm doing as is for now cuz each render takes about 30 min, I can remake
them on thursday

simpri21, 2015-12-02:
Saved what I have for now, will remake once everything's finalized

eric-wieser, 2015-12-02:
Did we get this on the dropbox?

simpri21, 2015-12-02:
If you haven't checked already, yes I saved it on there

eric-wieser, 2015-12-02:
Great

premphy, 2015-12-02:
Were you able to waterjet parts and trim rods?

________________________________
Chandan Sharma Subedi
Department of Mechanical Engineering
Massachusetts Institute of Technology
premphy@mit.edu | 617-608-8694