eric-wieser/2.12-robot #60: Implement arm torque control

eric-wieser, 2015-12-03:
Things to note:
* Overloads happen
* They can be detected with `motor.get_errors().Overload`
* They can be reset by setting `TorqueLimit` back to non-zero
* Doing this can cause overheats if done too often!

premphy, 2015-12-03:
Can someone open the door? Thanks

eric-wieser, 2015-12-03:
By which I mean set `tau = tau_pid + tau_static`

reaganm, 2015-12-03:
Dynamixel Conn got super fucked up and I don't know how to fix it. If you
have a copy of the correct code somewhere on your laptop, can you put it on
git under a new name?

eric-wieser, 2015-12-03:
`git checkout c77404105668dc7884b93296049e6c13349ce21f -- 'software/Dynamixel*.m'` will fix your problem

reaganm, 2015-12-03:
I did this but can't get anything

reaganm, 2015-12-03:
The dynamixel wizard won't allow me to reset the torque limit. how do i do
this?